RespiraWorks/SystemDesign #10: Humidifier housing design

neelfirst, 2020-04-15:
Polyimide heater + what kind of humidifier chamber, CPAP part good enough?

ebhillstrom, 2020-04-15:
Settling on a humidifier chamber will help us finalize the upstream hose connection. At this point, it seems like the humidifier chamber will be semi-disposable (not reused between patients). We should decide if the heater needs to be disposable as well. There may be a solution where heater is mounted into the enclosure with an exposed metal/metallic face and the humidifier sits on top of it like a hotplate? Finding existing ventilator design precedent would be useful here.

ebhillstrom, 2020-04-22:
Off the shelf humidifier heater will be used. Current baseline is HC150, which is $250.
[hc150-humidifier-cpap-fisher-paykel-user-manual.pdf](https://github.com/RespiraWorks/PhysicalDesign/files/4518609/hc150-humidifier-cpap-fisher-paykel-user-manual.pdf)

closed 2020-04-22 by ebhillstrom