RespiraWorks/SystemDesign #11: Select production-grade outlet solenoid / exhaust flow restriction

ebhillstrom, 2020-04-15:
--- From Ethan via Slack ---

This is a high flow valve. Specification is as follows:
- Pressure drop: 60 lpm through the valve with a pressure drop of less than 0.49 kpa (nominally a Cv of 2.3).
- Normally open is strongly preferable.
- Pressure is never more than 1.5 psi on either side.
- It needs to open at 0psig (direct lift).
- It doesn't need oxygen cleaning.
- 12V DC

We would plan to buy 3 to test and then ~100 as an initial run. Lead time obviously matters here. If there are any specifications that make this harder to source we are flexible. We'd consider two valves in parallel if that was required to get the flow/pressure drop with something in stock.

Three vendors we've tried:

1. Burkhart - pretty helpful, Colin Radmakker. Has sent us free valves.
2. Reinhold automation (MAC valves) - kind of helpful, no free valves, think they misunderstood the spec
3. US Solid - Quality/lifetime insufficient. Don't think we can put these in production hardware.

ebhillstrom, 2020-04-15:
One issue with alternative concept (even if effective, which is as yet unproven) is that we will need to qualify the valve ourselves for use in a ventilator, which likely means doing life cycle testing. If this is the route we go, it needs to be identified soon so that we can prioritize this.

ebhillstrom, 2020-04-15:
Further discussion of alternate valve concept to happen in issue #15.

esc-works, 2020-04-15:
Other options considred here have been a [banjo pinch valve](https://www.banjocorp.com/banjo/Valves/c/valves?q=%3Aname-asc%3Atype-facet%3APinch+Valve) or other [style pinch valves](https://www.clippard.com/products/isolation-valves-npp)

esc-works, 2020-04-16:
https://www.mksinst.com/f/t2b-ethercat-exhaust-throttle-valves?

esc-works, 2020-04-16:
The other holistic solution is to not worry about the exhale solenoid, but to just control the oxygen with a solenoid. See issue #4 and #14

esc-works, 2020-04-16:
I just bought some idle air control valves

esc-works, 2020-04-20:
Piloted mushroom valve?

ctung, 2020-04-22:
Looks like there are ventilator circuits you can get with exhalation valves built in: https://www.medonthego.com/Ventilator-Circuit-6-Ft-Adult-Dual-Limb-Nhtd-CareFusion-001793-1-Each_p_125090.html   I ordered a few for a different ventilator design.  I'll report on how they work if that would help.

neelfirst, 2020-05-05:
@sosaysthecaptain do you mind attaching your most recent solution to this ticket? Thanks.

sosaysthecaptain, 2020-05-05:
[image]

[exhaust pinch valve 1.4.step.zip](https://github.com/RespiraWorks/PhysicalDesign/files/4578385/exhaust.pinch.valve.1.4.step.zip)

[Onshape link](https://cad.onshape.com/documents/3fe0c1f79c482144c267173d/w/2ad1c08071a25185f9c78c68/e/fed9023927bfd7d2374ff967)

sosaysthecaptain, 2020-05-05:
Updated version: 
- dimensions adjusted for leverage
- stop is flat
- stop runs full length
- bearings better aligned

[image]
[image]
[exhaust pinch valve 1.5.step.zip](https://github.com/RespiraWorks/PhysicalDesign/files/4581141/exhaust.pinch.valve.1.5.step.zip)