RespiraWorks/SystemDesign #12: Decide on patient tubing connection standard

ebhillstrom, 2020-04-15:
Available dimensional standards: https://www.iso.org/obp/ui#iso:std:iso:5356:-1:ed-4:v1:en 15mm and 22mm are the two options in scope

Needs some research on common ventilator configurations and available breathing circuits on the market. The goal is to avoid cornering a hospital into needing to buy from a limited supply of breathing circuits. Note that the inspiratory connection should go to a humidifier, and _then_ to the patient, so the inspiratory connection should be compatible with a short hose that can jump to the humidifier inlet port - see issue #10.

neelfirst, 2020-04-15:
Just a comment for context - the humidifiers we've sourced for the first three units are CPAP parts and are thus 22mm compatible. I'm not sure what parts we can source on the cheap that would be 15mm compatible - I admit this is out of my expertise.

esc-works, 2020-04-15:
It seems like we should perhaps target medical device standard *in case* they have access to medical supplies, and in the lung run. If they are going to hack something together, we could instead also supply adapters. But I do not think we should design for non-standard hoses, if I understand here correctly.

ebhillstrom, 2020-04-15:
There is no suggestion of designing for non-standard hoses. It is simply that there are **two different standards** called out as acceptable in the ISO standard (see first line of thread): 15mm male cone and 22mm male cone. These get implemented non-uniformly in medical devices. For instance the Medtronic device has one 15mm male port and one 22mm male port, presumably intended to prevent interchanging the inspire and expire limbs. However, when I went "shopping" for compatible breathing circuits, I found suppliers that only supply circuits with 2x 22mm female connectors (e.g. https://www.draeger.com/Library/Content/accessory-ca-9105122-en-us-1910-2.pdf). There's a decision we need to make on what medical, ISO standard to conform with. I mostly want someone to take on doing a bit of the research on what's most commonly available to make that decision in an informed way.

ebhillstrom, 2020-04-15:
Possible correction to previous comment. I think the Medtronic design has the same size of connector, but one is male and one is female (still with the intent of discouraging flipping the connector ends, I presume). Just something we need to nail down and then move on with. See image:
[image]

esc-works, 2020-04-15:
Ok, so as I understand it:
- Current ventilator designs (a few we've reviewed) use a male 15mm conical and female 22mm. 
- The issue is in reviewing hose fittings, there are option for female/female and male/male hoses.
- So we need to figure out if there is a standard for the humidifier fittings to see if it is male/male or male female. 
- Looking at this humidifier, it is male/male (so 22mm/22mm hose) from the ventilator, and presumably female to the patient?
[image]

ebhillstrom, 2020-04-20:
Decision reached over Slack, and due to some research from Pallavi. We are going with 22mm male connections on the front face of the vent. All downstream connections to humidifiers and viral filters handled with female 22mm hose connections or adapters as needed. Relevant ISO standard dimensional info:

[image]
[image]

closed 2020-04-20 by ebhillstrom

neelfirst, 2020-04-23:
Slack reference: https://respiraworks.slack.com/archives/C0103PMM8DU/p1587404548098400