RespiraWorks/SystemDesign #14: Select oxygen control solenoid

ebhillstrom, 2020-04-15:
Not currently implemented in alpha. 

Requirements: 
- Normally Closed, Cv ~0.1-.2
- 50 psi
- ¼” NPT
- O2 cleaned/compliant

The O2 control valve is a binary (on/off) solenoid that gets a timing signal from the vent controller and is inside the vent. The O2 regulator is totally separate. It's possible we can get away without it.

Three vendors we've tried:

Burkhart - pretty helpful, Colin Radmakker. Has sent us free valves.
Reinhold automation (MAC valves) - kind of helpful, no free valves, think they misunderstood the spec
US Solid - Quality/lifetime insufficient. Don't think we can put these in production hardware.

esc-works, 2020-04-15:
Blocked by #4

esc-works, 2020-04-15:
@ngwood111@gmail.com: "As far as switching on/off the O2 supply later in the circuit, do not use solenoids that creates an explosion risk. O2 valves have to be very clean, no grease whatsoever, and I would do my best to avoid getting heat anywhere near O2 (and solenoids typically generate a fair bit of heat). I don't think people understand what pure oxygen does"

esc-works, 2020-04-21:
With no oxygen control, implemented this way, you can't control oxygen FiO2 to a very high value. This might not be acceptable

esc-works, 2020-05-15:
Either we're going to use a modified version of the [pinch valve](https://github.com/RespiraWorks/PhysicalDesign/tree/master/02_System_Design/pneumatic_design/pinch_valve) or we'll need to find an O2 compliant PSOL. [This](https://automationdistribution.com/smc-pvq31-6g-23-01n-pack-of-1/?gclid=CjwKCAjwkun1BRAIEiwA2mJRWdwnFDcsEDxhSQ3jvQuymCkzKmM2RFZsz48_6-lAaL0t2yUbvixLHxoCtrkQAvD_BwE) is the current leading contender. Two have been purchased and are en route.