RespiraWorks/SystemDesign #15: Test passive flow restriction valve

ebhillstrom, 2020-04-15:
Related to issue #11. Relevant comments:

https://github.com/RespiraWorks/PhysicalDesign/issues/11#issuecomment-614119995
https://github.com/RespiraWorks/PhysicalDesign/issues/11#issuecomment-614121020

This part is an alternative to an exhaust restriction solenoid, and is designed to minimize oxygen bypass and wastage. If effective it is much cheaper and saves supply chain lead time (could possibly be made in country of ventilator manufacture if tooling is available). However, it is a part that may need to be injection molded and which we will need to run qualification testing on ourselves to have design confidence.

sosaysthecaptain, 2020-04-16:
I'm running a second print of this tonight, having just mailed the first one to Bryan, lest it would be useful. Will test in the AM.

neelfirst, 2020-05-05:
@sosaysthecaptain do you mind attaching your most recent solution to this ticket? Thanks.

esc-works, 2020-05-15:
This was never tested appropriately, but modelica modeling indicated that leakage on the exhaust control valve was really problematic. We could re-approach this now that we have a faster response on the inhale, but the same solution for inhale seems applicable to exhale and provides more control.

closed 2020-05-15 by esc-works