RespiraWorks/SystemDesign #16: Non-return valve

esc-works, 2020-04-15:
Need a design/sourcing for non-return valve
[image]

esc-works, 2020-04-16:
Terence suggested: https://www.harvardapparatus.com/one-way-respiratory-valves.html#collapseTwo

neelfirst, 2020-04-16:
@tplusr ^^

esc-works, 2020-04-17:
@ebhillstrom suggested https://www.mcmaster.com/5492K13 
Working assumption for now

tplusr, 2020-04-17:
Should I order this part?

esc-works, 2020-04-21:
May need to re-open as those valves have a 0.33 psi (2.27527) cracking pressure. On the other hand that may work great for the blower

esc-works, 2020-05-15:
Unfortunately, this issue needs to be re-opened. The valve has a 0.3 psi cracking pressure, which is a little too high. It could still be used on the upstream leg but it makes the fan response worse. Recommendation was to look at this valve:   
[DU 210 004-154 XX.pdf](https://github.com/RespiraWorks/PhysicalDesign/files/4636042/DU.210.004-154.XX.pdf)