RespiraWorks/SystemDesign #4: Oxygen control valve

neelfirst, 2020-04-15:


esc-works, 2020-04-15:
[image]

esc-works, 2020-04-23:
Note to self to update this based on discussions with david wilkinson. High level planning to switch to an O2 compliant fan.

neelfirst, 2020-04-30:
Possible duplicate: #35

esc-works, 2020-05-07:
> Possible duplicate: #35

I think not a duplicate. We do have to check for O2 compliance in downstream parts though. Based on recent discussions, the answer here is going to be to provide dynamic O2 control, as doing anything with a static flow rate seems like a big lift with high uncertainty.