RespiraWorks/SystemDesign #6: Confirm filter housing

neelfirst, 2020-04-15:
- what kind of filters are we using (alpha: roomba hepa filters)
- does the filter housing fit the filters
- can the filter housing be mounted on the enclosure

ebhillstrom, 2020-04-15:
Current filter housing needs to be evaluated in the prototype to make sure there are no issues with recirculation. However, current baseline part fits Roomba HEPA filters and will be screwed onto the main enclosure through a rectangular cutout. Please advise if Roomba filter solution is not scalable, as we will need to revise before working with injection molding company.

ebhillstrom, 2020-04-15:
Seems like the Roomba filters have plenty available (and are cheap at scale!) but not sure on the lead time. https://www.aliexpress.com/item/4000002758100.html?spm=a2g0o.productlist.0.0.17a1f02dStzRtR&algo_pvid=28c860b1-92c2-403d-a0e0-c450135aa9bd&algo_expid=28c860b1-92c2-403d-a0e0-c450135aa9bd-13&btsid=0ab6f83115869732985805560ef19a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

ebhillstrom, 2020-04-22:
Now includes viral inline filtration as well, using these filters:

https://www.aliexpress.com/item/4000049950429.html?spm=a2g0o.productlist.0.0.3a4856881IvBih&algo_pvid=c2010458-dc02-4bed-be42-7e446a1a06a4&algo_expid=c2010458-dc02-4bed-be42-7e446a1a06a4-2&btsid=0ab50f0815875794215551884e430a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

ebhillstrom, 2020-04-22:
Response from a conversation on filtration system requirements:

------
What I know from other ventilator designs is the following:

1.   You can't rely on the end user always installing viral/bacterial filters in the patient circuit (including both the insp and exp limbs), therefore, the expiratory components are likely to get contaminated.  Therefore, these components must be sanitizable or disposable. 

2.  Additionally, the area where exhaled gasses leave the machine and enter the room should be kept far from any air intakes, user interfaces and any nearby surfaces should be easily cleanable

There may be more functional requirements for filtering in the 80601-2-12 standard or other standards.

The use of the Roomba HEPA filter seems like a clever idea.  The only concern I have is whether all of the materials in the filter are biocompatible. Do the filter membranes outgas, or leak anything into the airstream that could hurt the patient?  I'm not sure if Roomba HEPA filters are equivalent to "medical HEPA filters." Also, does the Roomba HEPA filter block virus-size particles from passing through?  Some research needs to be done here.

Sorry I don't have more concrete answers for you.  Bottom line is:

1.  I think your HEPA intake filter is a good start, just need to check that (a) it blocks viral/bacterial particles and (b) it is biocompatible
2.  Your expiratory path needs to be disposable or sanitizable

neelfirst, 2020-05-08:
@sosaysthecaptain @ebhillstrom this is done, right? we have filter housings designed around Roomba HEPA filters?

neelfirst, 2020-05-14:
Current solution is to have one mold with four filters: air intake/exhaust + patient inhale/exhale.

Open question - are the filter parts we've selected appropriate for viral filtering. How does this compare against ease of sourcing. This may require risk assessment if we downgrade e.g. to HEPA filters (0.3 um) 

We can expect individual manufacturers to come up with their solutions for substitutions. Thus we need to ensure size is something feasible, but there is no standard size to facilitate easy substitutions.

Need some standard plastic piece that molds to the filter and that we can seal. So this is a package deal we cannot be able to replace the filter.

sosaysthecaptain, 2020-06-23:
Here are files:

[image]

[filter parts.zip](https://github.com/RespiraWorks/SystemDesign/files/4822486/filter.parts.zip)
[image]
[image]
[image]
[image]

sosaysthecaptain, 2020-06-24:
These filter housings are to be used for both the inlet/outlet filters in-line filters. They accomodate the bare filter found in the HEPA filter catridge designed for the Roomba 700 series vacuum.

 The filter housing, patient connector, and filter hold-down are to be molded from ABS. Both gaskets are be cut from ~1/16" (1.5") rubber such as McMaster #1370N13. I'm assuming neoprene around a durometer of 60, although this is fairly flexible.

The filter housing is held down with 4x m4x8 SHCS, or longer based on the thickness of the panel. The halves are held together with m4x20 thumb screws, going into threaded inserts. I'm using heat-set for now; ultimately what we use will be determined after consultation with the molding engineers.

Note that the patient hose connector is made to fit a 22mm CPAP-type connector, and the internal connection is made to fit both 3/4" and 1" tubes.
--

1370N13
--

1370N13
--

1370N13
--

1370N13
--

1370N13
--

. I'm ass

[OnShape link](https://cad.onshape.com/documents/3fe0c1f79c482144c267173d/w/2ad1c08071a25185f9c78c68/e/11907480755add4f9ca862c4)

sosaysthecaptain, 2020-06-24:
Files: [filter parts.zip](https://github.com/RespiraWorks/SystemDesign/files/4822513/filter.parts.zip)