RespiraWorks/SystemDesign #8: Selection of overpressure relief valve

neelfirst, 2020-04-15:
On patient inhale side, after O2 mixing, before humidification.

TBD: before or after inhale pressure/flow measurement?

esc-works, 2020-04-15:
Yesterday we purchased a valve from McMaster Carr https://www.mcmaster.com/4277t52-4277T522

neelfirst, 2020-04-23:
Did we receive this valve and/or test it? Quick slack search doesn't turn anything up.

esc-works, 2020-04-23:
Yes, sorry! forgot to update. I think this is good as long as we have a fan solution (max flow rate is very low but will reliably prevent high pressures as long as its not like 100 cm H2O). Believe we can close this.

neelfirst, 2020-04-24:
Confirmed part https://www.mcmaster.com/4277t52-4277T522, closing

esc-works, 2020-05-15:
This issue is re-opened. Testing the mcmaster part revealed the first one was leaking at about 20-25 cm of pressure. It also had a very small orifice that would have not allowed for real relief anyway. This is annoying but we'll need a better solution.

dtgerson, 2020-05-22:
We also have to figure out what pressure we want this to relieve at

dtgerson, 2020-05-28:
see this chain for some valves we decided to try out: https://respiraworks.slack.com/archives/C0137BLF17A/p1590501836181400