RespiraWorks/SystemDesign #9: Finalize flow measurement solution

neelfirst, 2020-04-15:
dP - what kind of chamber

vs

MAF or other sensor paradigm

esc-works, 2020-04-15:
Flow sensor specification (note this is the physical interface, assuming the pressure sensor @Edwin Chiu has selected).

- Reading range: 0-80 lpm (@John Batteh whats the largest flow we're seeing on the inflow side?)
- Signal strength: differential pressure range must be between 0-4 kPa
- Pressure loss: no more than 0.1 kPa at 60 liters per minute
- Manufacturing: Should be manufacture-able from injection molded parts (can be two piece). Can be 3d printed for development. Can include machined/stamped parts.
- Pressure sensor can mount to flow instrument but the baseline uses an NXP dP sensor mounted to the PCB with tubing.
- Must interface with 2.5 mm ID tubing (current flow sensor)
- The error/accuracy must be sufficient that during exhale, when the fan is driving flow through both ports, the signal between inhale flow rate and exhale flow rate is distinguishable.
- Response time must be on the order of 50 ms

esc-works, 2020-04-21:
On suggestion from RED team, may consider these
[DataSheet_PMF4000_MassAirFlow_RevE_C1.pdf](https://github.com/RespiraWorks/PhysicalDesign/files/4511991/DataSheet_PMF4000_MassAirFlow_RevE_C1.pdf)

Also, this link had math for the venturi design: https://docs.google.com/spreadsheets/d/1G9Kb-ImlluK8MOx-ce2rlHUBnTOtAFQvKjjs1bEhlpM/edit#gid=963553579

esc-works, 2020-04-21:
Also, thought is that the issue with the venturi above related to leakage rather than any issue with the sensor

neelfirst, 2020-04-23:
@esc-works Is this actually done?

esc-works, 2020-04-23:
I think we're getting close. It seems like for alpha/beta we'll go with a venturi design. Roughly a 20 degree entrance and 10 degree exhale, with a 4.8mm throat. Its tough because this is almost no resolution around 1 lpm. But it kind of works. We need to characterize it. This is the go forward path until we learn something else

neelfirst, 2020-04-23:
Let's keep this open then to track the characterization effort of the leading solution. Thanks!

neelfirst, 2020-04-24:
Merging #3 into this ticket.

neelfirst, 2020-05-05:
Can we get a condensed summary of testing for this ticket. If the Venturi design as specified above is no longer deemed the best approach why is this and what is the new approach. Thanks

neelfirst, 2020-05-14:
@esc-works i can haz data

esc-works, 2020-05-15:
I think this is still the go forward solution as nothing else we've identified has been better. We've tested and will continue to refine the accuracy. It works well with the cheap pressure sensors. We'll find a way to manufacture it. I think we close this ticket as it shouldn't be actively worked on until we learn something else.

closed 2020-05-15 by esc-works

jlebar, 2020-05-18:
Some traces of pressure/flow/etc are available at https://github.com/RespiraWorks/VentilatorSoftware/tree/master/sample-data