RespiraWorks/SystemDesign #30: Spec out sw dev care package

neelfirst, 2020-04-23:
Create with part numbers and links a package to essential kit a developer needs to move forward with testing against hardware.

neelfirst, 2020-04-23:
Linux SBC: https://www.mouser.com/ProductDetail/Raspberry-Pi/RPI4-MODBP-2GB-BULK
uC Module: https://www.mouser.com/ProductDetail/STMicroelectronics/NUCLEO-L452RE
7" touchscreen: https://www.aliexpress.com/item/4000747984746.html
HDMI-uHDMI cable: https://www.amazon.com/Rankie-Micro-Supports-Ethernet-Return/dp/B00Z07JYLE/
micro USB cable: https://www.amazon.com/dp/B07JBN6C5C/
jumper wires: https://www.amazon.com/GenBasic-Piece-Female-Jumper-Wires/dp/B01L5ULRUA/

neelfirst, 2020-04-23:
Reference to other components. 

Warning: this BOM is live - https://docs.google.com/spreadsheets/d/1mW5Jr8qKsH8XmUAp5w34JR3UQW4Njy3H/edit#gid=2003192236

closed 2020-04-24 by neelfirst

martukas, 2020-04-29:
Where are the assembly instructions for this?