RespiraWorks/SystemDesign #34: spike: research how to validate blower in 100% o2

neelfirst, 2020-04-30:
We could save ourselves some trouble by just proving our blower works in a 100% o2 environment (or not). How do we do this?

esc-works, 2020-05-07:
Not quite. From discussion with two manufacturers its really about UL certification and fire risk. Seems just way too risky to say "mfgr says you can't run O2 through this, but we think they are wrong and will do it anyway."

esc-works, 2020-05-15:
I propose to just close this. We're not going to blow O2 through the fan it seems like... Ugh but with the pinch valve after the fan it _would_ be really useful. Maybe we approach a vendor to see what this would take. We have contacts with Moog but they were the ones who said "no way, no how."

closed 2020-06-25 by martukas