RespiraWorks/SystemDesign #37: Design battery backup circuit

neelfirst, 2020-05-05:
There is definitely a reference to battery backup circuitry. Presumably we are using 12V SLAs but deciding where in the circuit they go and how to handle power crossover are open items.

inceptionev, 2020-05-29:
I have been mostly convinced so far that we should use an external AC UPS, because (1) they are complete systems with integrated alarms and self test features and (2) because it would avoid the extra certification and safety burden of bringing in mains AC voltage into the unit, and also certifying the backup system function.

inceptionev, 2020-05-29:
However, I have looked a bit into what it would take to bring a battery backup inside the unit, and I believe this is the easiest and most straightforward path: 

1. an AC-DC supply with an integrated sealed lead-acid charging function (such as MeanWell DRC-100A).
2. a 12V 7-9Ah sealed lead acid battery (such as B&B BP7-12-T2‎)
3. a filtered and fused input module (such as Schurter DD14-52020-12111-1100‎) and the associated fuse holder (Schurter 4301.1403‎) and fuses (Schurter ‎0034.3120‎)
4. the ability to measure the DC input voltage (already supported on PCB)
5. add an input to the PCB that can observe the AC OK signal from the AC power supply.

here's a digikey cart that parts this out for ~$130: https://www.digikey.com/short/zmpwtz