RespiraWorks/SystemDesign #56: FiO2 sensor selection

neelfirst, 2020-05-17:
Right now we are stuck with galvanic cell O2 sensing which is more widely available but degrades quickly (12 months or faster if we use it in closed loop O2 control).

We would like to switch to oxygen fluorescence quenching but it is currently a supply chain risk.

neelfirst, 2020-05-17:
From @inceptionev 

> The oxygen sensor I was using was SST LOX-02, which it looks like I remembered incorrectly as being able to support 100% O2 sensing, according to the datasheet, it is 25% max unfortuantely

inceptionev, 2020-05-28:
We have purchased a few of these for evaluation: https://www.cablesandsensors.com/products/compatible-o2-cell-for-hamilton-medical-396008

These are common medical electrochemical oxygen sensors.  There are many types that are all very similar, so if we don't end up using this exact one, we will be able to get something very similar to it.  I.e. a 13mV ambient electrochemical oxygen cell with ~12month service life.