RespiraWorks/SystemDesign #57: Volume Calculation

dtgerson, 2020-05-17:
This issue is meant to cover the issues/thoughts around estimation of volume.

Related issues: #17 and (software needs to be implemented) https://github.com/RespiraWorks/VentilatorSoftware/issues/50

High level: tidal volume is not directly measured but is calculated based on other sensors.

Current design: Venturi 1 (using a single DP sensor) before patient and Venturi 2 (using a single DP sensor) after patient both measure flow rate. A single pressure sensor exists between the two (right before Venturi 2). No temperature or humidity measurements.

Volume is then calculated by measuring the flow through Venturi 1 and subtracting out the flow through Venturi 2 and integrating this over time to calculate the amount of volume delivered to the patient.

Currently there is a challenge in getting accurate volume measurements. As can be seen in this post, the volume measurement seems to drift steadily - rather than integrating out to around zero for most breaths. See this post for example data: https://respiraworks.slack.com/archives/C011ABVBS21/p1589590403271900?thread_ts=1589583371.264500&cid=C011ABVBS21

This behavior seems to be driven by noise/drift from the DP sensors - causing the flow measurements to be off, driving the volume measurement to be off. We believe that for our particular DP sensors, the offset is very temperature dependent.

Ideas to investigate for improving this:
1. Add processing in software to compensate 
--Could we estimate the linear drift and then subtract this out? Or could we reset the volume to 0 after each breath (because theoretically the patient should exhale all of the volume)? Or could we use a filter that incorporates information about the expected behavior?
--Concerns with this approach: If the volume measurement is sufficiently off, it seems like this is problematic for estimating volume of a single breath (breaths that are too big and breaths that are too small are both problematic). There are also concerns with breath stacking and being able to catch problematic trends across breaths. But we still don't fully understand this requirement that well.

2. Add more DP sensors and then integrate the readings - or get better DP sensors

3. Add temperature sensing/humidity sensing to the DP sensors and use that to compensate the measurements

4. Change sensors (e.g. mass flow sensors? - Edwin has more info)

5. Do more characterization of the DP sensors to understand their drift/offsets

Related open questions:
1. What resolution do we need on this volume measurement?

jlebar, 2020-05-17:
> This behavior seems to be driven by noise/drift from the DP sensors

Specifically it's drift.  There is also a lot of noise in the sensors, but we have enough time to take lots of samples and average out the noise, no prob.  But when the noise average *changes*, and changes very quickly like it did in these examples (average went from -1.2L to 0L basically instantaneously), that's where we get into trouble.

neelfirst, 2020-05-17:
Follow-up from #9

jlebar, 2020-05-18:
Some traces of pressure/flow/etc are available at https://github.com/RespiraWorks/VentilatorSoftware/tree/master/sample-data.  I am not sure if these drift, though.

esc-works, 2020-05-18:
@ebhillstrom: 

> - Datasheet says 1.5% maximum error for 0 to 100 mm H2O - this is usually % full scale, making the error +/- 1.5mmH2O
> - @jlebar data for the "zero" test has all measured values below .2mmH2O - it now makes sense that the offsets are different because they're all well within sensor precision - you won't get better measurements than that no matter what you do. If we plotted it out, the random drift is also probably well within this accuracy meaning you probably won't be able to correct it.
> - From the venturi spreadsheet, 1.5mmH2O (the sensor accuracy) corresponds to ~7lpm, which is a lot! Plus this error gets doubled when you subtract two dP measurements.
> - From this it seems like the sensor isn't fine enough (or the orifice is too big) to measure low flows."

esc-works, 2020-05-18:
- We could zero/bring very low the flow "every breath" if we don't _always_ have bias flow.
- We could schedule the "inhale" valve to fully close right at the beginning of expiratory flow. (If there is leakage I think we could characterize the leakage and subtract it out but zero is an easier value to know.) Even just a few miliseconds here would help. 
- Similarly on the expiratory side right at the start of inhale we could close the exhale valve. That should temporarily zero the exhale flow.
- we can't do this currently because there was no way to 'zero' the inhale flow. but with the valve we can. I wonder if  we could try this on one of the solenoid pizzas just on the exhale side. It would limit the walk of one sensor at least which should cut down on the overall walk.

neelfirst, 2020-05-18:
Datasheet: https://datasheet.octopart.com/MPXV5004DP-Freescale-Semiconductor-datasheet-12520.pdf

Autozero app note 1: https://www.nxp.com/docs/en/application-note/AN1636.pdf

Autozero app note 2 (not the one I posted last night): https://sensing.honeywell.com/auto-zero-calibration-technique-pressure-sensors-technical-note.pdf

neelfirst, 2020-05-18:
In particular, that 1.5% is _with_ regular autozero calibration!

asmodai27, 2020-05-19:
> There is also a lot of noise in the sensors, but we have enough time to take lots of samples and average out the noise, no prob

This is something we are not doing for now, but should be pretty straightforward, both with oversampling and filtering.
Said filtering will be much easier now we have a timer interrupt that allows us to have constant time between samples, though it requires setting the timer interrupt on a higher frequency than it currently is (100 Hz) and running the PID calculations every Nth interrupt.

jlebar, 2020-05-19:
>> There is also a lot of noise in the sensors, but we have enough time to take lots of samples and average out the noise, no prob

> This is something we are not doing for now,

45e4242344d046df61d94d120a32950a0dab0112 landed, so we are oversampling and averaging (without any filtering).

esc-works, 2020-05-24:
I wanted to take a minute to try and re-specify the "problem statement" here, so we know what "done" is. We are measuring flow and integrating two flow sensors to get volume. 
So the first requirement is flow. I'm going to propose a "success" criteria/tests for flow as follows:
- **Flow Accuracy** be within +/- 2 liter per minute at 5 lpm, and +/- 1 liter per minute at 30 and 80 liters per minute of flow, on both inspiration and expiration, when differenced between the two limbs.
-**BTPS** Incorporate BTPS correction to the exhale volume. https://d3cumwfln92soa.cloudfront.net/8a716ac3fd123ce0becd7b56596582d4fc4c0c47/appnote-btps-correction-v01r.pdf
-**Volume accuracy.** I propose that we specify that the drift induced integration error is small relative to our leak detection requirements. The accuracy should be within 10% of the smallest tidal volume we expect to measure over the longest breath we expect to measure. A relatively low-range minute ventilation of 5 liters per minute is taken as a baseline (standard is 6-8, or 20% less than that for **4 lpm**. While the device is specified down to 5 bpm, that should only ever occur for either a) an incredibly fit person breathing 800mL/breath or b) a patient spontaneously breathing on SIMV. If we assume 10 bpm as the minimum fully supported breathing rate, then the breath occurs over 6 s. At a minute ventilation of 4 lpm, the tidal volume is 400 ml. For a 10% accuracy, the correction factor being applied to "zero" the flow can be no larger than 400ml/6s or 67 ml/s or 6.7 cL/s.

bruprescott, 2020-05-28:
The use of an SFM3300 proximal sensor could solve this issue.

Is the implementation viable?

jlebar, 2020-05-28:
I believe the main concern is cost.

bruprescott, 2020-05-28:
The cost is not that high.
Two DP sensors are the value of that sensor.

https://br.mouser.com/ProductDetail/Sensirion/SFM3300-D?qs=u16ybLDytRa%2FjhQAbeBcMw%3D%3D

asmodai27, 2020-05-28:
"Disposable Single Use Version" not sure what that really means, but would it need to be replaced every so often?
Also, it looks like we are on the right path to make the volume estimation work with the current sensors (see #flow_measurement on slack). Need some more characterization and testing but we have outlined some solutions that work on paper.

esc-works, 2020-05-28:
Yeah, the issue is not cost, we had evaluated those and couldn't get them in any quantity, ("This part number is not currently available from Mouser. Product may be in limited distribution or a factory special order.") and they cannot be sterilized so need to be replaced with every patient ("Disposable Single Use"). So the cost for one is pretty comparable, but the Venturi can hopefully be sterilized.

bruprescott, 2020-05-29:
This restriction warning appears, but I just bought 5 units to test.

I'll see how the circuit is set up. We can evaluate a way to sterilize and not just use one per patient.
The AW version can be sterilized, but it costs more than 100USD.

I have tested the venturi with this team's code, but I find many error values at low volume.
In all the tests I saw using the SFM3300, they were super stable and it is not necessary to recalibrate.

esc-works, 2020-05-29:
@bruprescott cool! thanks. please let us know how it goes. @inceptionev we tested one of these once, right? Do you still have it?

bruprescott, 2020-05-31:
Has anyone tested automotive MAF sensors?

I found an approach on the subject.
https://github.com/hydronics2/COVID-19-Airflow-Sensor-AFH55M12

asmodai27, 2020-06-04:
My attempts at building an algorithm that can take care of the flow bias and drift of our current flow sensor finally yield very promising results. Many parameters will need tuning and testing on actual hardware (which I will try as soon as I have a full pizza setup), and here it goes:

For now, this needs to have 0 flow at some point, but I think I can come up with a similar solution when the flow is constant but not 0:
- build rolling average of pressure delta signal over a reasonably long period (in my tests: 0.2s, long enough to average the noise pretty well but short enough to not filter too much of the actual dynamic of the system).
- integrate flow with a 2L/min deadband (corresponding to the sensor noise level).

Every so often (for implementation convenience I am using the same duration as the rolling average window: 0.2s), re-zero the sensor if both of these are within reasonable bounds meaning they are caused by noise and drift, not an actual signal:
- average pressure delta < time since last auto-zero * max allowable drift (meaning that the current average value can be attributed to drift)
- integral of flow since last check < max volume for auto-zero (indicating the averaged pressure delta  isn't hiding an actual small flow: because of the noise, a constant 1L/min flow leads to a measurable integral of flow during these 0.2s, even with a 2L/min deadband. This also prevents re-zeroing when the average pressure delta is hiding a dynamic signal, which also leads to a significant flow integral).

Results of this algorithm on synthetic pressure signals, built from sample data available on VentilatorSoftware repo, simulating 100 s "vent off" followed by twenty 5s breaths (built from 2020-05-14-pip25-peep10-rr12-ie23.csv). noise and drift are built to match values from 2020-05-14-vent-off.csv and my own characterization of the sensors on non-zero flow noise level.
[image]

On the flow:
[image]

For the volume integration, I am also using the 2L/min deadband to get rid of the noise and so long as the re-zeroing algorithm is working properly, this gives me very good results (in the simulations I am running, accurate to within 50 mL over 5 minutes, re-zeroing every 10 breaths brings it back with ease to within 5 mL and shouldn't cause too much trouble).

Results for the volume:
[image]

Octave scripts that allow me to run these (renamed as .txt so github will let me attach them):

[PressureDeltaToFlow.txt](https://github.com/RespiraWorks/SystemDesign/files/4732753/PressureDeltaToFlow.txt)
[RollingAverage.txt](https://github.com/RespiraWorks/SystemDesign/files/4732754/RollingAverage.txt)
[volume_simulation.txt](https://github.com/RespiraWorks/SystemDesign/files/4732755/volume_simulation.txt)

asmodai27, 2020-06-07:
Results with the actual ventilator software available on slack, similar to the ones here. No volume measurement for now since I don't have a full setup and the only thing I can do is feed flow to the venturi.