RespiraWorks/SystemDesign #58: Ideas for Integration Testing (to verify the system - not just manufacturing)

dtgerson, 2020-05-17:
Running list of ideas for integration testing

dtgerson, 2020-05-17:
Things that need to be tested:
1. Combination of different PIP, PEEP, RR settings, etc.
2. Compare volume estimates to known volumes
3. O2 reservoirs of different back pressures
4. Different FiO2 settings

martukas, 2020-05-17:
Pressure sensor validation across a range of pressures, as compared to calibrated pressure sensor.

dtgerson, 2020-05-17:
validation of all sensors across a range of the things they are measuring against calibrated versions of those sensors

switching between various settings while running (e.g. changing PIP, PEEP, FiO2, RR, etc.)

testing alarms and different failure modes to see if the system reacts as expected

martukas, 2020-05-17:
validation of multiple sensors across a multidimensional space of changing parameters simultaneously? :)

martukas, 2020-05-17:
Validation of vertical integration of stepper control from software to actuation.
Similar tests for fan, solenoid, whatever other actuators are present.

rwilbur, 2020-05-22:
long term stability test of the system (make sure the pi doesn't crash, etc)

martukas, 2020-05-23:
Effect of humidity and temperture on flow sensors.