RespiraWorks/SystemDesign #85: Valve coefficient for pinch valves

esc-works, 2020-05-27:
[image]
One thing we need is to better characterize the response of the pinch valve. I am starting to plan a test to get better data on the pinch valve response curve because we'll need this. Unfortunately we need to measure pressures larger than 4kPa, so we'll need some new sensors to get this data. [These might work](https://www.digikey.com/product-detail/en/nxp-usa-inc/MPX5050DP/MPX5050DP-ND/464057). Anyway, the idea would be to capture various flow rates by turning the fan speed through a full range (slowly) at a number of different openings. We should then be able to back out a loss coefficient as a functioning of opening. I'll be working on this but if others were looking for something to do. I need to build a pinchy valve

inceptionev, 2020-05-30:
Performed Pinch Valve characterization at blower power setting of 200 out of 255 full scale.  At full power, the dP sensors saturate and do not give useful readings for significant portions of the valve range.

Setup photo:

[image]

Data Plot:

[image]

Link to data:

https://drive.google.com/file/d/1oJH50d26ScL5DKg482lFd2sMBVO-UwtM/view?usp=sharing

brmurray, 2020-05-31:
[also posted in #integration_testing]
I took a stab a fitting curves to the full pinch valve test data you posted.
Version 2 removes the valve positions beyond which the flow is totally restricted, and so yields nicer s-shaped fit. Version 1 has all the valve positions for completeness.
Python script also included.

[image]
[image]