RespiraWorks/SystemDesign #88: Oxygen FiO2: Ambient air PEEP in patient hose will be delivered instead of FiO2 mix

inceptionev, 2020-05-28:
Current CONOPs has oxygen inlet valve closed during exhale to conserve oxygen, using ambient air instead to provide PEEP.  This will purge the patient tubing with ambient air.  If the tubing is long and has significant volume, this is the gas that will be delivered to the patient when the next inhale cycle begins, regardless of the oxygen mix in the ventilator.

Potential solution is to maintain FiO2 mix during PEEP.  This wastes more oxygen but guarantees an FiO2 mix in the patient tubing at all times.

neelfirst, 2020-05-29:
Admittedly my conversations have been with Western doctors, but the input I've received so far is that they would prioritize accuracy in O2 measurement and delivery, and then secondarily focus on conserving O2 as a resource.

esc-works, 2020-05-29:
My feeling is strongly this: We need to focus on getting FiO2 correct. It's mostly a control algorithm change to implement. Therefore we should not change the air mixture during exhale, we should maintain the FiO2 for both inhale and exhale. Note that this may mean we want to control the exhale valve during exhale to let the patient exhale generate most of the pressure needed.