RespiraWorks/SystemDesign #90: Equipment selection for integration testing

martukas, 2020-05-29:


martukas, 2020-05-29:
Ken Krieger's suggestion is to get ourselves a National Instruments suite, buy a LabView license and have our test suites run on that. All sensors can be analog. Here are his suggestions:

https://www.ebay.com/itm/National-Instruments-Vision-Development-Module-License-For-Labview-CD-2014/303579931654
https://www.ebay.com/itm/NATIONAL-INSTRUMENTS-MOD-FP-D0-400-8-CH-DISCRETE-INPUT-W-MODEL-FP-TB-1-T10/362935228030
https://www.ebay.com/itm/National-Instruments-NI-FP-DI-330-8-Ch-Discrete-Input-Module-WITH-BASE-FP-TB-1/192449378989
https://www.ebay.com/itm/NATIONAL-INSTRUMENTS-FP-AI-110-8-CH-16-BIT-ANALOG-INPUT-J1/402152868926
https://www.ebay.com/itm/NATIONAL-INSTRUMENTS-FP-AI-110-8-CH-16-BIT-ANALOG-INPUT-Y2/192998706640
https://www.ebay.com/itm/NATIONAL-INSTRUMENTS-FP-AI-110-8-CH-16-BIT-ANALOG-INPUT-Y3/401818956182
https://www.ebay.com/itm/National-Instruments-FP-DI-300-8-CH-Discrete-Input-Excellent-Used-Guaranteed/382414348393
https://www.ebay.com/itm/National-Instruments-FP-AI-110-Analog-Input-FieldPoint/262495254460
https://www.ebay.com/itm/National-Instruments-FP-TB-1-FieldPoint-Terminal-Base-184106C-01/143612964427
https://www.ebay.com/itm/National-Instruments-FP-1000-RS-232-RS-485-Network-Interface-184120E-01/293229536416
https://www.ebay.com/itm/NATIONAL-INSTRUMENTS-10-100-MBPS-ETHERNET-NETWORK-INTERFACE-FP-1600/324107390406

martukas, 2020-05-29:
All hands meeting on May 29 resulted in quick decision to go with this:
https://www.digikey.com/product-detail/en/labjack-corporation/T7/2173-T7-ND/10447019
https://www.digikey.com/product-detail/en/labjack-corporation/CB37/2173-CB37-ND/10447027

martukas, 2020-05-29:
Still need somewhat confident decisions on the following:
* pressure sensor
* flow sensor
* pot on stepper (measure stepper movement directly)
* any intermediate device needed to measure hall effect pulse from blower?
* microphone for alarm sounds?
* serial dongles for comms testing?

martukas, 2020-05-29:
What machine do we run the jenkins node on? Is rpi sufficient?

esc-works, 2020-05-29:
1x these is on the way to you: https://store.ncd.io/product/ams5915-0100-d-amplified-low-pressure-sensor-0-100-mbar-0-to-1-450-psi-i2c-mini-module/

esc-works, 2020-05-29:
I also don't think we need to measure fan speed. If anything we can measure fan power/current pretty easily if you want.

martukas, 2020-05-29:
I think there is still debate on whether we might want to measure fan speed. It is not that much more complex to implement this and will be less uncertainty in signal. We'll have channels plenty to DAQ this.

martukas, 2020-05-29:
Webcam to observe pizzas remotely? Any insight from buying it on aliexpress?

martukas, 2020-05-29:
Sound like current meter on fan power line is a (perhaps better) alternative to reading hall effect sensor.