RespiraWorks/SystemDesign #108: CVD - PCB "Custom Components" Section

esc-works, 2020-06-08:
Pointer to PCBreathe
Any assembly instructions

https://docs.google.com/document/d/1wSYof_d_WnFvfAGxjNq9vY3AhY8o4VpuVuHDv58NDv8/edit#heading=h.cicffg7tb2l6

inceptionev, 2020-06-11:
Here's a link to the complete turnkey manufacturing package.  You should be able to hand this package of documents to any standard PCB fabricator and assembly shop and have them produce fully-assembled boards.  This includes the Bill of Materials, Fabrication Drawing, Assembly Drawing, Pick-and-Place centroid file, PCB gerber artwork, and Schematic.  https://github.com/RespiraWorks/pcbreathe/blob/master/NIGHTLY/20200424v2-RELEASE-CANDIDATE-2/20200424v2-RespiraWorks-Ventilator-Rev1.0-RC2-PKG-TURNKEY.zip

inceptionev, 2020-06-16:
Updated PCB section in Custom Components doc with description and links to github design and manufacturing packages.

closed 2020-06-19 by esc-works