RespiraWorks/SystemDesign #110: CVD - Alarm: Volume

esc-works, 2020-06-08:
Download the “decibel X” app. Place a phone 1 meter from ventilator. Measure a baseline decibel output. 
Trigger an alarm and record the output. 
Goal: 60 to 80 dBa at one meter

sglow, 2020-06-19:
OK, I just tested this on the controller using the buzzer on that board.

Here's a screen shot showing the buzzer going full blast in an otherwise quiet room.  I tweaked it a bit in software so now I'm getting about 70dB

[image]

sglow, 2020-06-19:
When the blower is running full blast it's already at almost 60dB.  An alarm at that point raises the din up to about 75dB
[image]

sglow, 2020-06-19:
Note that these measurements were for the buzzer on the controller only.  I'm not sure if that's really the alarm for the vent since most of the alarm stuff is being handled on the GUI.  Someone who has a working GUI setup should test the volume of the alarm that the touch screen speakers produces

esc-works, 2020-06-20:
Not perfect but I'm going to close the issue and move on. Thanks @sglow !

esc-works, 2020-06-20:
@jkff wants to update this. @martukas maybe has some data

inceptionev, 2020-06-21:
Got the GUI up and running and filmed the testing of the volume.  Alarm testing video here: https://youtu.be/_GUU8BG_yBc

Screenshot here: 
[image]

Side note: the app that CoVent asks you to download is a $5/month subscription ::blech::

jkff, 2020-06-21:
Section "SR-25 Alarm Volume shall be 60 to 80 dBA at one meter (+/- 5 dBA)" in Performance Evaluation Report now covers this.

closed 2020-06-21 by jkff