RespiraWorks/SystemDesign #116: CVD - Pressure Compatibility Testing

esc-works, 2020-06-08:
Hook up ventilator to regulator with >50 psi supply pressure (pretending this is hospital O2).
Take a picture.

esc-works, 2020-06-16:
we're probably not going to get to this test since we haven't picked a regulator

jkff, 2020-06-20:
@inceptionev plans to clarify with @dtgerson exactly what should be done here.

dtgerson, 2020-06-20:
if I'm understanding what this is about, I would say edwin should just take a picture of his ox tank with the reg hooked up to the system (ideally in a way that shows the pressure of the ox tank) and then we can post that and say we're making progress

inceptionev, 2020-06-21:
there is a video and test data for pressure compatibility now here: https://github.com/RespiraWorks/VentilatorSoftware/tree/inceptionev/20200621-OxygenPressureCompatibilityTest/sample-data/20200621-OxygenPressureCompatibilityTest

inceptionev, 2020-06-21:
^ the above runs test #6 with pressure from 100psi down to 20psi

inceptionev, 2020-06-21:
we still need someone to do the write-up

closed 2020-06-25 by martukas