RespiraWorks/SystemDesign #117: CVD - Respiratory Rate Range

esc-works, 2020-06-08:
Screenshot from UI
Documentation of software
If time is available, video or trace of BPM clicking through from 10-30 bpm with a low compliance/lowest resistance lung

jlebar, 2020-06-12:
@esc-works can you confirm that what you're looking for here is that the ventilator won't let you use a respiratory rate outside of 10-30bpm?

@paulovap how hard would this be to implement in the GUI?

esc-works, 2020-06-12:
correct @jlebar. Basically enforcing a range of 10-30 for command breaths. I think the only real "testing" we would need to do is (in order of complexity)
1: show the relevant software requirement
2: simulate the UI (in a browser, for example) and show the range is enforced.
3: hook up the ventilator to the test lung and demonstrate each of the RRs.

paulovap, 2020-06-14:
Not sure if I understood correctly. Is that the video you need?

https://drive.google.com/file/d/17TjqfHmuqaNzXMhAAoR5kYd517DpmFEv/view?usp=sharing

See also:
https://github.com/RespiraWorks/VentilatorSoftware/pull/512

esc-works, 2020-06-14:
> RespiraWorks/VentilatorSoftware#512

This is perfect.
-Ethan

esc-works, 2020-06-16:
Test request, relatively low priority, would be choosing some setting back in the ventilator table settings and stepping through each respiratory rate just showing that in fact the ventilator can respond that quickly. Alternatively (and perhaps better) is to define the rise time and fall time required for hitting a RR = 30 condition and recording a test demonstrating those.

esc-works, 2020-06-20:
RR test, use the low, medium, high limits and record 2 breaths at each. 
10, 20, 30 bpm. Use whatever lung settings make this test easiest.

**Approximate Testing Procedure**

1. Set up testing configuration (connect pressure sensor to the ventilator output, just before entry into the test lung)
2. Start pressure and flow data logging
3. Turn on the ventilator
4. Set the desired Respiratory Rate
5. Run ventilator for data collection duration
6. Repeat steps 4-5 for each PEEP setting (10, 20, 30)
7. Turn off ventilator
8. Stop data logging

inceptionev, 2020-06-21:
Done as above

inceptionev, 2020-06-21:
Respiratory Rate demo data, video and photos are up here: https://github.com/RespiraWorks/VentilatorSoftware/tree/inceptionev/2020-06-21-RespiratoryRateDemo/sample-data/2020-06-21-RespiratoryRateDemo

closed 2020-06-22 by esc-works

jkff, 2020-06-22:
Respective section of Performance Evaluation Report also filled in now.