RespiraWorks/SystemDesign #122: CVD - Export Zip Altium Package

esc-works, 2020-06-08:


inceptionev, 2020-06-16:
Design Files: https://drive.google.com/file/d/1bujMaEhlp7HlqBhJcG2bO4sFTghJyMfL/view?usp=sharing

Manufacturing Files: https://drive.google.com/file/d/13IxR1hi02mj8zuQjK6Vfy6ekf4aIQ03Q/view?usp=sharing

esc-works, 2020-06-16:
Woo!! first thing done. FYI I renamed these files and moved them to the "ready for submission" folder

inceptionev, 2020-06-16:
Actually, probably better to use github links since we are open-source:

Design: https://github.com/RespiraWorks/pcbreathe/blob/master/NIGHTLY/20200424v2-RELEASE-CANDIDATE-2/20200424v2-RespiraWorks-Ventilator-Rev1.0-RC2-PKG-DESIGN.zip

Manufacturing: https://github.com/RespiraWorks/pcbreathe/blob/master/NIGHTLY/20200424v2-RELEASE-CANDIDATE-2/20200424v2-RespiraWorks-Ventilator-Rev1.0-RC2-PKG-TURNKEY.zip

closed 2020-06-19 by esc-works