RespiraWorks/SystemDesign #124: CVD - System, Hardware and Software Requirements

esc-works, 2020-06-08:
Export the RW-SYS requirements from Valispace

neelfirst, 2020-06-13:
[rw-sys-reqs-2020-06-12-nks.pdf](https://github.com/RespiraWorks/SystemDesign/files/4774147/rw-sys-reqs-2020-06-12-nks.pdf)

neelfirst, 2020-06-13:
@esc-works a quick review if you please. moving to hw/sw reqs next.

esc-works, 2020-06-14:
This looks good to me!

neelfirst, 2020-06-17:
[rw-hw-reqs-2020-06-16-nks.pdf](https://github.com/RespiraWorks/SystemDesign/files/4790435/rw-hw-reqs-2020-06-16-nks.pdf)

neelfirst, 2020-06-17:
@esc-works @dtgerson @paulgun89 hardware requirements up, review requested

neelfirst, 2020-06-17:
note I've hidden the dependencies here, but you can view them in valispace and/or the matrix i'm on the hook for later.

neelfirst, 2020-06-17:
The more I work on these the more I realize we will need to do some serious upkeep post-covent. On my list to find some dedicated requirements engineers.

neelfirst, 2020-06-19:
I'm updating the SW reqs a bit and finding more and more that we need to overhaul all of our subsystem requirements. The system requirements are the most presentable, of course, but I don't think I can bring the HW/SW reqs up to the same level on time.

neelfirst, 2020-06-19:
Submitted: https://drive.google.com/file/d/1pktSzkpa1WcbzBCZT6jgiz5fpdssPRS3/view?usp=sharing

closed 2020-06-19 by neelfirst