RespiraWorks/SystemDesign #141: CVD - AD Instruments Testing

esc-works, 2020-06-16:
This is testing for compliance with ISO 80601-2-12 Table 201.104 and 201.105 (note we were only testing 201.105). 
We DO NOT have to test all lines. Just a representative sample.

For each test:
Write the test setup down, include pictures if possible
We record the native export file from the LabChart program. For each test we need to report:
Lung Settings
Compliance
Resistance
Ventilator Settings
TV (if volume control)
PIP (if pressure control)
Set rate
I_time
FiO2
PEEP
Measured results
Measured TV
Measured PIP 
Peak flow rate (LPM)
Measured RR
Measured I_time
Measured O2% (if recorded)
Measured PEEP

inceptionev, 2020-06-21:
Run-for-Record data is here https://github.com/RespiraWorks/VentilatorSoftware/tree/inceptionev/20200620-AIR-tuning-update/sample-data/20200621-RunForRecord1-AIR-and-OXY

inceptionev, 2020-06-21:
Video here: https://youtu.be/tyYFPTh4At0

inceptionev, 2020-06-21:
Note that OXYGEN Test #4 was done at the wrong lung settings, so there is a redo ventilator data test 4 file and 9 runs in the ADI test data instead of 8.  If time permits I will do a clean new oxygen run of all 8 tests.

inceptionev, 2020-06-21:
New complete oxygen dataset uploaded

inceptionev, 2020-06-21:
If we do a second run-for-record, here is a wish-list:
- ADI flow autozero off
- Maybe 5 more seconds of test data per line
- Add calculated volume to data collection

inceptionev, 2020-06-21:
New data set with the above wish list is here: https://github.com/RespiraWorks/VentilatorSoftware/tree/inceptionev/20200621-RunForRecord2/sample-data/2020-06-21-RunForRecord2-AIR-and-OXY

esc-works, 2020-06-21:
all the data (raw acdicht export) we need gets uploaded to the covent submission folder: https://drive.google.com/drive/u/1/folders/1o4Kr_9_fYovlg636wIGeBFWF83NUe6za
the filled out excel table
all files renamed appropriately
identified in 00 cover page
called out in 02-1 SR-04