RespiraWorks/SystemDesign #142: CVD - Breath Detection

esc-works, 2020-06-16:


esc-works, 2020-06-19:
Clarified this issue above as being about assist control, and added a new ticket for pressure support. 

It looks like this ticket testing is more or less complete as of today. Ideal data would be a waveform showing the inspiratory effort, and then the delivered breath. In the interim I will use this youtube video: https://www.youtube.com/watch?v=PXKGaeBhqVI&feature=youtu.be

inceptionev, 2020-06-21:
Breath Detection Walkthrough is up: https://www.youtube.com/watch?v=6vUZYqXCQ2s  Data, video, and photos are here: https://github.com/RespiraWorks/VentilatorSoftware/tree/2020-06-21-BreathDetectionWalkthrough/sample-data/2020-06-21-BreathDetectionWalkthrough

jkff, 2020-06-21:
This all appears to be currently documented under https://docs.google.com/document/d/1g7qLD5qD4BKfR1mcGq7-QY6XE2C9oIIw5GJdUzU31Zg/edit#heading=h.pt7ef8fp4ywf.

jkff, 2020-06-21:
Is there anything more remaining on this issue or should we mark it done?

closed 2020-06-22 by jlebar